GuNanOvO/openwrt-tailscale #6: Issue with install on OpenWrt 24.10.0  MediaTek MT7621 ver:1 eco:3 Netgear R6350

Walleye0072, 2025-05-23:
Tailscale service does not start properly. Below is what I see after I start the service and from the log file. 

root@OpenWrt:~# service tailscale start
logtail started
Program starting: v1.84.0-t0b36774ff, Go 1.24.3: []string{"/usr/bin/tailscaled", "--cleanup"}
LogID: 97a0487cf85f78e2f7f5e7adb98e7d16da85def0e2135dbc60d48de416e2001e
logpolicy: using system state directory "/var/lib/tailscale"
dns: [rc=unknown ret=direct]
dns: using "direct" mode
dns: using *dns.directManager
dns: inotify: NewDirWatcher: context canceled
flushing log.
logger closing down
root@OpenWrt:~# cat  /var/lib/tailscale/tailscaled.log.conf
{
        "Collection": "tailnode.log.tailscale.io",
        "PrivateID": "20224141b0b6e963f22766f3186086ccac5d6c531221cd17e8cb786a6ee42662",
        "PublicID": "97a0487cf85f78e2f7f5e7adb98e7d16da85def0e2135dbc60d48de416e2001e"
}
root@OpenWrt:~#

I installed with the script on this page: 
https://github.com/GuNanOvO/openwrt-tailscale/blob/main/README_en.md

tailscale works from the command prompt but since the service is not running, it will not do anything. 

Thank you!!!!

Pete.

GuNanOvO, 2025-05-24:
Hi @Walleye0072, thanks for reporting!

Can you try the following steps to help us diagnose the issue?

Run:

`uname -m `

Start the service manually:

`/etc/init.d/tailscale start `

Then run:

`tailscale up`

Let me know what you see after each command.

Thanks!

Walleye0072, 2025-05-24:
Here is the output:BusyBox v1.36.1 (2025-02-03 23:09:37 UTC) built-in shell (ash)
 
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.0, r28427-6df0e3d02a
 -----------------------------------------------------
***@***.***:~# uname -m
mips
***@***.***:~# /etc/init.d/tailscale start
logtail started
Program starting: v1.84.0-t0b36774ff, Go 1.24.3: []string{"/usr/bin/tailscaled", "--cleanup"}
LogID: 97a0487cf85f78e2f7f5e7adb98e7d16da85def0e2135dbc60d48de416e2001e
logpolicy: using system state directory "/var/lib/tailscale"
dns: [rc=unknown ret=direct]
dns: using "direct" mode
dns: using *dns.directManager
dns: inotify: NewDirWatcher: context canceled
flushing log.
logger closing down
***@***.***:~# tailscale up
failed to connect to local tailscaled; it doesn't appear to be running
***@***.***:~#
 Cheers,Peter ***@***.*** from Peter Dyk's mobile.

GuNanOvO, 2025-05-24:
Hi @Walleye0072,

Thanks for taking the time to dig into this and sharing the details! Could you please try the following steps?

1. First, reboot your device and then try running `tailscale up` again to see if it works.
2. If it still doesn’t work, try reinstalling using the script:
    1. Uninstall tailscale completely.
    2. Make sure you haven’t installed tailscale via opkg before (if you did, remove it).
    3. Then run the install script again from scratch.

I don’t have a real mipsle device, but I tested it on a QEMU virtual machine with the same mipsle architecture, and everything worked fine — I was able to run `tailscale up` with no problems. So this issue you’re having is really strange.

Let me know how it goes!

Walleye0072, 2025-05-24:
### Results after reboot: 
root@OpenWrt:~ tailscale up
failed to connect to local tailscaled; it doesn't appear to be running
root@OpenWrt:~
I did try to install tailscale from okpg. It indicated not enough memory. I check installed packages and tailscale does not show up. 

I reinstalled with the script. I see there is an error with:
Collected errors:
 * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /lib/libustream-ssl.so
        But that file is already provided by package  * libustream-mbedtls20201210
 * opkg_install_cmd: Cannot install package libustream-openssl.

[OpenWRT_ssh_install_text.txt](https://github.com/user-attachments/files/20427141/OpenWRT_ssh_install_text.txt)

Walleye0072, 2025-05-24:
I checked to see what ssl packages are installed. Below is the result: 
root@OpenWrt:\~\# opkg list-installed | grep -i ssl
libopenssl3 - 3.0.16-r1
luci-ssl - 24.328.43591\~9e70c9a
root@OpenWrt:\~\#

GuNanOvO, 2025-05-24:
Hi, thanks again for spending time on this!

Could you try this:

1. Run `/etc/init.d/tailscale start`
2. Then run `ps | grep tailscale` to check if `tailscaled` is actually running
3. If not, run '/etc/init.d/tailscale start && logread -f' to start the service and watch logs in real time.
4. Wait a bit to catch enough logs

Let me know what shows up, especially in the logs — this might help pinpoint what’s going wrong.

GuNanOvO, 2025-05-24:
This issue is really strange. Could you please try running tailscaled manually like this:

'/usr/bin/tailscaled --state=/etc/tailscale/tailscaled.state'

Then run:

'dmesg'

and paste the full output here. We shouldn’t miss any detail.

Walleye0072, 2025-05-24:
Here are the results of the two above commands. See the log files. One is logread after I ran /etc/init.d/tailscale start. The second is from the execution of /usr/bin/tailscaled --state=/etc/tailscale/tailscaled.state. 

Below is the output of dmesg command. It would seem that running tailscale a second time is causing an out of memory kill of the tailscale process. 

root@OpenWrt:\~\#

[usr_bin_tailscaled command output.txt](https://github.com/user-attachments/files/20427764/usr_bin_tailscaled.command.output.txt)
[logread _ grep tails.txt](https://github.com/user-attachments/files/20427765/logread._.grep.tails.txt)

# dmesg | tail -n50
[10824.336452] Mem-Info:
[10824.341086] active_anon:11304 inactive_anon:3753 isolated_anon:0
[10824.341086]  active_file:0 inactive_file:126 isolated_file:0
[10824.341086]  unevictable:0 dirty:0 writeback:0
[10824.341086]  slab_reclaimable:596 slab_unreclaimable:5129
[10824.341086]  mapped:3416 shmem:7470 pagetables:149
[10824.341086]  sec_pagetables:0 bounce:0
[10824.341086]  kernel_misc_reclaimable:0
[10824.341086]  free:3940 free_pcp:31 free_cma:0
[10824.417030] Node 0 active_anon:15012kB inactive_anon:13828kB active_file:476kB inactive_file:24kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:13664kB dirty:0kB writeback:0kB shmem:29880kB writeback_tmp:0kB kernel_stack:936kB pagetables:596kB sec_pagetables:0kB all_unreclaimable? no
[10824.469788] Normal free:15760kB boost:0kB min:16384kB low:20480kB high:24576kB reserved_highatomic:0KB active_anon:15012kB inactive_anon:45216kB active_file:496kB inactive_file:128kB unevictable:0kB writepending:0kB present:131072kB managed:119264kB mlocked:0kB bounce:0kB free_pcp:132kB local_pcp:0kB free_cma:0kB
[10824.525095] lowmem_reserve[]: 0 0
[10824.531839] Normal: 259*4kB (UME) 139*8kB (UME) 51*16kB (UME) 67*32kB (UME) 31*64kB (UME) 21*128kB (UME) 3*256kB (ME) 2*512kB (U) 1*1024kB (U) 0*2048kB 1*4096kB (M) 0*8192kB = 16692kB
[10824.564649] 7545 total pagecache pages
[10824.572164] 0 pages in swap cache
[10824.578805] Free swap  = 0kB
[10824.584554] Total swap = 0kB
[10824.590319] 32768 pages RAM
[10824.595870] 0 pages HighMem/MovableOnly
[10824.603527] 2952 pages reserved
[10824.609827] Tasks state (memory values in pages):
[10824.619239] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
[10824.636449] [    881]    81   881      389        0    16384        0             0 ubusd
[10824.652775] [    882]     0   882      272       32    20480        0             0 askfirst
[10824.669608] [    917]     0   917      310       32    20480        0             0 urngd
[10824.685934] [   1345]   514  1345      572       32    20480        0             0 logd
[10824.702095] [   1399]     0  1399      901       64    20480        0             0 rpcd
[10824.718293] [   1699]     0  1699      326       32    16384        0             0 dropbear
[10824.735157] [   1805]     0  1805      735       32    16384        0             0 hostapd
[10824.751881] [   1806]     0  1806      735        0    20480        0             0 wpa_supplicant
[10824.769782] [   1833]   101  1833     1169       96    24576        0             0 wpa_supplicant
[10824.787666] [   1834]   101  1834     1179       96    20480        0             0 hostapd
[10824.804337] [   1870]     0  1870      692       96    16384        0             0 netifd
[10824.820869] [   1987]     0  1987      477       64    20480        0             0 odhcpd
[10824.837405] [   2109]     0  2109      765       64    24576        0             0 uhttpd
[10824.853924] [   2524]     0  2524      350       32    20480        0             0 udhcpc
[10824.870423] [   2525]     0  2525      312        0    20480        0             0 odhcp6c
[10824.887107] [   3079]     0  3079      735       32    20480        0             0 dnsmasq
[10824.903792] [   3098]   453  3098      429        0    16384        0             0 dnsmasq
[10824.920519] [   3373]     0  3373      735       31    16384        0             0 ntpd
[10824.936720] [   3424]   123  3424      350       32    20480        0             0 ntpd
[10824.952916] [   3458]     0  3458      275       64    16384        0             0 ubihealthd
[10824.970140] [  22770]     0 22770      331       64    20480        0             0 dropbear
[10824.987000] [  22780]     0 22780      351        0    20480        0             0 ash
[10825.003010] [  25396]     0 25396   139629     9282    94208        0             0 tailscaled
[10825.020245] [  25679]     0 25679      331        0    20480        0             0 dropbear
[10825.037110] [  25680]     0 25680      351        0    16384        0             0 ash
[10825.053096] [  25736]     0 25736     7774      672    28672        0             0 tailscale
[10825.070103] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=tailscaled,pid=25396,uid=0
[10825.095957] Out of memory: Killed process 25396 (tailscaled) total-vm:558516kB, anon-rss:24064kB, file-rss:0kB, shmem-rss:13064kB, UID:0 pgtables:92kB oom_score_adj:0

Walleye0072, 2025-05-24:
Here is the tailscale executable: 
root@OpenWrt:\~\# ls -h -l /usr/bin/tailscale*
lrwxrwxrwx    1 root     root          19 May 24 12:05 /usr/bin/tailscale -> /usr/bin/tailscaled
-rwxr-xr-x    1 root     root        5.5M May 24 12:05 /usr/bin/tailscaled
root@OpenWrt:\~\#

GuNanOvO, 2025-05-24:
Hi, thanks a lot for testing and sharing all the details.

I’m really sorry, Tailscale seems a bit too heavy for your device’s limited memory. Even though we compressed the binary to save space, that doesn’t help with the memory it needs while running.

You might want to try [Zerotier](https://openwrt.org/docs/guide-user/services/vpn/zerotier) as an alternative, it usually uses less memory and could work better on small OpenWrt devices like yours.

Hope ZeroTier works out well for you! Good luck!

GuNanOvO, 2025-05-24:
Thanks again for your patience and the time you spent testing this — really appreciate it!

I’m very sorry that there's not much we can do about the out-of-memory issue on your device due to its limited RAM. You can now safely run the script again to uninstall Tailscale and clean things up.

Also, just a heads-up: since one of your comments included IP address info in the logs, I’ve deleted it to protect your privacy.

Wishing you all the best — take care!

Walleye0072, 2025-05-24:
I ran the top command. I then ran /usr/bin/tailscaled --state=/etc/tailscale/tailscaled.state

What does the 467% of memory mean? 

<samp>Mem: 98252K used, 21012K free, 15696K shrd, 0K buff, 26308K cached
CPU:   2% usr   3% sys   0% nic  93% idle   0% io   0% irq   0% sirq
Load average: 0.15 0.05 0.17 2/114 690
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  535 25680 root     S     545m 467%   1% /usr/bin/tailscaled --state=/etc/tailscale/tailscaled.state
  405 22780 root     R     1416   1%   0% top
<samp/>

Walleye0072, 2025-05-24:
Also, how much memory should this skinny version of tailscale use?

GuNanOvO, 2025-05-25:
Yeah, that 467% memory number from top does look crazy — but don’t worry, it’s not actually using 467% of your memory. On OpenWrt (with BusyBox), those numbers can be pretty unreliable or just plain wrong.

Just to give you an idea: I’ve got a device with 256MB of RAM, and even the stripped-down version of Tailscale uses over 60MB while running — that’s already taking up most of the available memory. So for a smaller device like yours, it's not surprising that tailscaled gets killed due to running out of memory.

Tailscale is really heavy for these low-end setups, even after compressing the binary.

winkmal, 2025-10-25:
Thanks for your great work on bringing Tailscale to more OpenWrt routers.
> `failed to connect to local tailscaled; it doesn't appear to be running`

Same error here on another MediaTek MT7621 device with 128 GB RAM and 16 MB storage, running current OpenWrt 24.10.4  
First, I ran into the `iptables` issue. Then, I tried [your suggestion](https://github.com/GuNanOvO/openwrt-tailscale/issues/26#issuecomment-3415988068) to `opkg install iptables-nft ip6tables-nft`
But it still throws lots of errors, and LuCI shows 82 % of RAM being used. 😦