GuNanOvO/openwrt-tailscale #17: 关于内存占用

bowen-shiyong752, 2025-07-20:
非常感谢GuNanOvO为我们提供这么好的openwrt插件实现vpn应用。现在我的 q30pro 512内存使用手动持久安装tailscaled-linux-arm-normal后使用非常稳定，开机内存260m左右。但是链接远程webdav后满速下载（32MB/s左右）约2分钟后会内存溢出死机、重启。是否可以对tailscale限速或者限制内存使用解决？谢谢！

GuNanOvO, 2025-07-20:
可以尝试修改`/etc/init.d/tailscale`:
```bash
vi /etc/init.d/tailscale
```
当中的`procd_set_param env TS_DEBUG_FIREWALL_MODE="$fw_mode"`一行，在后面加上`GOGC=10`参数  ，`procd_set_param env TS_DEBUG_FIREWALL_MODE="$fw_mode" GOGC=10`提高内存回收频率。
然后重启再试试看，可能会提高cpu占用。
来自reddit的[相关讨论](https://www.reddit.com/r/Tailscale/comments/1htinza/tailscale_on_openwrt_memorydisk_usage/)。

GuNanOvO, 2025-07-20:
如果可以的话，反馈一下效果，添加参数后的运行内存占用，以及cpu占用，我参考一下，后续加到文档里面，不胜感激。

bowen-shiyong752, 2025-07-20:
添加参数后，不会死机重启了。cpu96-100占用，剩余内存14左右。十分感谢！

closed 2025-07-20 by GuNanOvO

GuNanOvO, 2025-07-20:
在tailscale官方库的issues里有一个相关问题[https://github.com/tailscale/tailscale/issues/7272#comment-composer-heading](https://github.com/tailscale/tailscale/issues/7272#comment-composer-heading)，一直没有得到解决，我也查看了一下，似乎没有更好的解决方法。可以尝试用其他插件限制一下速度，也许有用？实在没办法的话，换成zerotier、netbird这些同类型的服务，应该要比tailscale情况好些。

GuNanOvO, 2025-07-21:
对，继续加到上面的gogc后面

bowen-shiyong752, 2025-07-21:
好的，谢谢！我试试！

bowen-shiyong752, 2025-07-22:
50太少了，速度很慢。加到128，能跑100兆，cpu占用90。十分感谢！

GuNanOvO, 2025-07-22:
好的，感谢反馈❤️主要是观察稳定性，GOMEMLIMIT参数加上之后能够稳定使用的话，我继续把它加到文档里面供大家参考。关于进程的问题我不太清楚，应该是正常的。

bowen-shiyong752, 2025-07-22:
好的，十分感谢！