GuNanOvO/openwrt-tailscale #23: MIPS architecture

Merith-TK, 2025-08-09:
[image]
```
Package tailscale version 1.86.4 has no valid architecture, ignoring.
# uname -a
Linux OpenWrt 6.6.93 #0 SMP Mon Jun 23 20:40:36 2025 mips GNU/Linux
```

Latest Firmware for my router, just updated today

Google Translate English to Chinese
谷歌英语翻译成中文

`软件包 tailscale 版本 1.86.4 没有有效的架构，忽略。`
我的路由器的最新固件，今天刚刚更新

GuNanOvO, 2025-08-09:
Hi,
The repository does contain the package for the correct architecture (`mipsel_24kc`), so based on the current information I’m not sure where the issue is coming from.

Please try running:

```sh
opkg update
```

and then attempt the installation again.

Alternatively, you can install it directly via URL to bypass the package index:

```sh
opkg install https://gunanovo.github.io/openwrt-tailscale/tailscale_v1.86.4_mipsel_24kc.ipk
```

GuNanOvO, 2025-08-09:
Glad to hear it worked!
Out of curiosity, what exactly did you try that made it work — was it running `opkg update` first, or installing directly via the URL?

Merith-TK, 2025-08-09:
I just installed directly via the URL, could be something to take note of!

GuNanOvO, 2025-08-09:
Sorry to be a bit pedantic, but I’d like to confirm — did you skip trying `opkg update` followed by `opkg install` and only use the direct URL install? This will help me better understand where the root cause might be.

Merith-TK, 2025-08-09:
okay, to take it from the top
1) ran the instructions to install tailscale
2) got error
3) opened this issue
4) `opkg install <direct url>`
5) this current conversation

also i just tried
```
root@OpenWrt:~# opkg find $(which tailscale)
Package tailscale version 1.86.4 has no valid architecture, ignoring.
Package tailscale version 1.86.4 has no valid architecture, ignoring.
Package tailscale version 1.86.4 has no valid architecture, ignoring.
(repeat for a total of 64 times)
```

winkmal, 2025-10-25:
> was it running `opkg update` first, or installing directly via the URL?

On a MIPS device running the current OpenWrt 24.10.4, the `opkg update` actually succeeded. But not without the abovementioned warning.

```bash
root@OpenWrt:~# opkg install tailscale
Package tailscale version 1.88.4 has no valid architecture, ignoring.
#...
Package tailscale version 1.88.4 has no valid architecture, ignoring.
Installing tailscale (1.88.4) to root...
Downloading https://gunanovo.github.io/openwrt-tailscale/tailscale_v1.88.4_mipsel_24kc.ipk
Installing kmod-tun (6.6.110-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.4/targets/ramips/mt7621/kmods/6.6.110-1-ae7dcdf01cb63b83c64f5cb8d8960511/kmod-tun_6.6.110-r1_mipsel_24kc.ipk
Configuring kmod-tun.
Configuring tailscale.
logtail started
```

winkmal, 2025-10-25:
Thanks for the prompt reply.

> If you can run `tailscale --version` or `tailscale up`, it means it was installed successfully.
```bash
root@OpenWrt:~# tailscale --version
1.88.4
  tailscale commit: 85f4267ee3bcf6a4e6b6d4384b66ee4cb33f1434
  long version: 1.88.4-t85f4267ee
  go version: go1.25.3
```

It first _seemed_ to run properly, but I cannot connect to Tailscale. See [this post](https://github.com/GuNanOvO/openwrt-tailscale/issues/6#issuecomment-3446650676) for more details.

GuNanOvO, 2025-10-25:
*> Thanks for the prompt reply.
> ```bash
> root@OpenWrt:~# tailscale --version
> 1.88.4
>   tailscale commit: 85f4267ee3bcf6a4e6b6d4384b66ee4cb33f1434
>   long version: 1.88.4-t85f4267ee
>   go version: go1.25.3
> ```
>
> It first _seemed_ to run [...]

Try running:
`/etc/init.d/tailscale start`
This should start the tailscaled service.

winkmal, 2025-10-29:
As an additional note: I tried installing the package on another OpenWrt device with ipq40xx platform. It also complains several times about `no valid architecture`. Then, I get the iptables error again. Any chance this is going to be fixed soon? Maybe you could add this to the installation guide, since it seems to occur on every installation!?
```bash
Installing tailscale (1.90.4) to root...
Downloading https://gunanovo.github.io/openwrt-tailscale/tailscale_v1.90.4_arm_cortex-a7_neon-vfpv4.ipk
Installing kmod-tun (6.6.110-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.4/targets/ipq40xx/generic/kmods/6.6.110-1-eaef302ef5ab82928154706763925f63/kmod-tun_6.6.110-r1_arm_cortex-a7_neon-vfpv4.ipk
Configuring kmod-tun.
Configuring tailscale.
logtail started
Program starting: v1.90.4-t68cba300e, Go 1.25.3: []string{"/usr/bin/tailscaled", "--cleanup"}
LogID: fbc6f174593585f48c78743eafbcd1edd35add8e7ae7d752d83faafefb70f941
logpolicy: using system state directory "/var/lib/tailscale"
logpolicy.ConfigFromFile /var/lib/tailscale/tailscaled.log.conf: open /var/lib/tailscale/tailscaled.log.conf: no such file or directory
logpolicy.Config.Validate for /var/lib/tailscale/tailscaled.log.conf: config is nil
dns: [rc=unknown ret=direct]
dns: using "direct" mode
dns: using *dns.directManager
linuxfw: clear iptables: exec: "iptables": executable file not found in $PATH
linuxfw: clear ip6tables: exec: "ip6tables": executable file not found in $PATH
dns: inotify: NewDirWatcher: context canceled
flushing log.
logger closing down
```